thijse/Arduino-CmdMessenger #3: 2 usage questions

orgicus, 2015-07-15:
Hi,

Sorry to use the issues to ask questions, but I couldn't find a group/forum for this library.
I have started using the library it looks great!

I did manage to get confused about 2 things:
1. How can I check how many arguments are present in a call ? In a scenario where a variable number of arguments is passed, how can I could the arguments ? (I've noticed that trying to read an argument that isn't there there returns 0.0 for readFloatArg() for example. Is there something I might have missed that already handles argument count ?)
2. I haven't managed to properly use the SendAndReceiveBinaryArguments function. All I could get the sketch to react to was sending a 0;  as binary (0x00 0x3B). I don't think I understood how the binary communication works. Hints ? :)

Thank you,
George

thijse, 2015-08-25:
Hi, 

Pleased to hear that you like the library! We very are interested in hearing what you are using it for.

1.  Under .NET this is possible: ReceivedCommand.Arguments.Count will give you the number of parameters. On your arduino, this is not know: to preserve precious memory, there is no read-ahead, and parameters are directly read from the stream. If you want, you can easily send the number of parameters as the first parameter.
2. You mean the sample is not working? Are you sure speed & port are correct, 16/32 bit is set correctly, etc?

thijse, 2015-08-25:
Ps. The best place to post questions is here: http://forum.arduino.cc/index.php?topic=186928.0

closed 2015-08-25 by thijse