thijse/Arduino-CmdMessenger #4: Reading a string messes up all the arguments.

Davideddu, 2015-08-29:
I've been using CmdMessenger for a while and it seemed to work, however my current application requires the reception of a string through serial. Here's the function:

```c++
void onBtChName() {
  sendAck(kACK, kBT_CHNAME, F("Storing new name."));
  uint8_t len = constrain(cmdMessenger.readInt16Arg(), 0, 30);
  cmdMessenger.copyStringArg(bt_newname, len+1);
  Serial.print("New name ");
  Serial.write(bt_newname, len);
  Serial.println(len);
  bt_newnamelen = len;
  bt_chname = true;
  sendAck(kDONE, kBT_CHNAME, F("New name stored. Please commit."));
}
```

This is the callback attached to kBT_CHNAME, which is command 15. I'd expect it to read two arguments, an integer and a string. However it does not work. If I comment out the string reading part, everything works fine (ex. if I send `15,12,34,56,78;`, the `len` variable would contain 12 as I would expect). However, if I also read the string (which is odd, as the string is read *after* the integer), it skips two arguments. So while `15,9,Davideddu;` does not work, `15,0,0,9,Davideddu;` does. Another example, if I send `15,12,34,56,78;` len would be set to 56 (actually 30 because of constrain, but it doesn't really matter) and bt_newname would contain 78 + 28 bytes of crap.

This is a pretty odd bug.

P.S.: I wrote a python port of CmdMessenger with a similar API that, however, tries to make use of Python-specific features. you can find it [here](https://github.com/Davideddu/python-cmdmessenger). I'll document it and add a readme at some point.

thijse, 2015-08-31:
Hmm, this is very strange. If you can me provide a compilable sketch that displays this behavior, I will check it out!

Davideddu, 2015-09-01:
This one: https://github.com/Davideddu/BluEOS-Arduino
Should work on ATmega328 or any other mcu with more flash/sram.
Uncomment the 2nd and the 3rd lines to enable the bluetooth API, then send

```
15,<length>,<string>;
 ```

for example

```
15,7,Arduino;
```

then try one of these instead:

```
15,0,0,7,Arduino
15,this isn't,being read,7,Arduino;
```

The first one will print this:

```
3,15,Storing new name.;
New name 0
12,15,New name stored. Please commit.;
```

However what it should print (which is what you get with the second command) is this:

```
3,15,Storing new name.;
New name Arduino7
12,15,New name stored. Please commit.;
```

If you run `14;` (kCOMMIT) right after that, it should print "AT+NAMEArduino" (where Arduino is whatever string you sent earlier) then restart.

Davideddu, 2015-09-01:
P.S.: you need [SoftReset](https://github.com/WickedDevice/SoftReset).
`15` is `kBT_CHNAME` which accepts two arguments, an integer (the length) and a string (the new name which will be sent to the bluetooth serial module.

ilovetogetspamed, 2016-02-02:
I'm having similar issues with cmdMessenger and readStringArg() via serial port, not bluetooth.  Any time I send from Python using either function, the Arduino stops responding to other commands being sent.  Any help would be greatly appreciated.

Note: 27 is a status message sent by the AVR, 31 is a comment sent by the AVR for debugging.
#### Log:

27,2,0,0,,0.00,0.00;
31,Entered: request_machine_id;
31,Entered: request_machine_mfg_info;
31,Entered: request_machine_mfg_contact;
31,starting Genie;
31,resetting LCD Screen;
31,Initializing RFID;
31,Reached loggedInUserType switch default;
31,AVR received_machine_id;
31,AVR received_head_count;
31,AVR received_needle_count;
31,received_machine_mfg_info;
  
#### Note I never get the received_machine_mfg_contact message

If I remove the two functions which use cmdMessenger.readString  I don't have any issues.  Of course that doesn't help since my applications needs those two functions to work.

### The Arduino Mega2560 functions to receive data
```
...
void receive_machine_mfg_info()
{
    char * manufacturer = cmdMessenger.readStringArg();
    char * model = cmdMessenger.readStringArg();
    char * serial_number = cmdMessenger.readStringArg();
    char * date_purchased = cmdMessenger.readStringArg();
    
    //genie.WriteStr(uLCD_STRINGS7, manufacturer);
    //genie.WriteStr(uLCD_STRINGS8, model);
    //genie.WriteStr(uLCD_STRINGS9, serial_number);
    //genie.WriteStr(uLCD_STRINGS10, date_purchased);

    cmdMessenger.sendCmd(k_echo_message, F("received_machine_mfg_info"));
}
void request_machine_mfg_contact()
{
    cmdMessenger.sendCmd(k_echo_message, F("Entered: request_machine_mfg_contact"));
    cmdMessenger.sendCmd(k_request_machine_mfg_contact);
}
void receive_machine_mfg_contact()
{
    char * contact_info_1 = cmdMessenger.readStringArg();
    char * contact_info_2 = cmdMessenger.readStringArg();
    char * contact_info_3 = cmdMessenger.readStringArg();

    //genie.WriteStr(uLCD_STRINGS11, contact_info_1);
    //genie.WriteStr(uLCD_STRINGS12, contact_info_2);
    //genie.WriteStr(uLCD_STRINGS13, contact_info_3);
    
    cmdMessenger.sendCmd(k_echo_message, F("received_machine_mfg_contact"));
}
...
```
### the Python code that sends the data

```
...
    @staticmethod
    def request_machine_mfg_info(self, *args):
        return '9,{0},{1},{2},{3};'.format(
            config().manufacturer,
            config().model,
            config().serial_number,
            config().date_purchased
        )

    @staticmethod
    def request_machine_mfg_contact(self, *args):
        return '11,{0},{1},{2};'.format(
            config().contact_info_1,
            config().contact_info_2,
            config().contact_info_3
        )
...
```

### Note: the conf().* comes from a separate Python class.

ilovetogetspamed, 2016-02-02:
I think I figured it out.  Instead of reading the string, comma's (,) and all.  It seems to me it's looking for more args after the commas instead of gobbling up the entire string. If I send:
<b>"9,ACME,Roadrunner Trap,203321,19 FEB 1999;"</b>
the function attached to command 9 only sees the 'ACME' part when it does the readString(), anything after the comma isn't read and the system 'hangs' there, presumably waiting on more args.

valkuc, 2016-05-25:
You have an error in your code in onBtChName method. Line:
`uint8_t len = constrain(cmdMessenger.readInt16Arg(), 0, 30);`
constrain is a macro:
`#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))`
so in your case "amt" parameter get expanded three times and when code is running "cmdMessenger.readInt16Arg()" got called three times...

To solve you problem just assign cmdMessenger call to temporary variable:
```
uint8_t len = cmdMessenger.readInt16Arg();
len = constrain(len, 0, 30);
```

closed 2016-05-25 by valkuc