thijse/Arduino-CmdMessenger #5: Nuget package

BytesGuy, 2015-09-07:
Hey Thijs, thanks for the great library! I was trying to implement virtually the same thing myself, but it was taking too long and had some nasty bugs. Stumbling upon this has got me up and running once again with minimal effort.

I have created some nuget packages for CmdMessenger to make it easier to install and use for the PC side of things - just wanted to know if you are happy for me to publish and maintain them on nuget.org

I have split the bluetooth transport layer into it's own package as it has a dependency (just keeps the core package nice and lean).

hunyadizsolt, 2015-10-16:
Yes very interested. Pleaseeeee. Thank you

thijse, 2015-10-19:
Awesome! Having Nuget packages would be great, and I am not at all opposed to somebody else doing all the work in maintaining them :-). I will contact you to discuss this.

closed 2015-10-19 by thijse

Rast1234, 2016-12-24:
Excuse me, i can't find the CmdMessenger package on NuGet. Can you post a link?