thijse/Arduino-CmdMessenger #8: New Python Interface

harmsm, 2016-05-25:
Hi Thijs,

Thanks for the great library.  I've found it quite useful.  

I've written a Python interface for an arduino running the CmdMessenger library called PyCmdMessenger (https://github.com/harmsm/PyCmdMessenger).  I built this specifically for communication between a raspberry pi and arduino over USB. 

Before I put the Python code onto Pip, I wanted to check with you and make sure you're fine with my use of the PyCmdMessenger name.  Also, if you like the Python interface, it might also be worth linking to my project from the main CmdMessenger page.  There are a huge number of python-centric raspberry pi users that would find the Arduino-CmdMessenger + PyCmdMessenger combination useful.  

Thanks,

Mike

thijse, 2016-06-01:
Hi Mike,

Great work! I have absolutely no problem with you naming it PyCmdMessenger. I will add a link on the CmdMessenger page. I have no knowledge of Python, but am I correct that it more resembles the basic, single thread implementation of the Arduino code, than the more complex C# variant?

Thijs

harmsm, 2016-06-04:
Hi Thijs,

Great, thanks!  I'm finding your library quite useful for a robotics
project I'm working on.

Yes, the implementation is a simple, single thread.  (If a user wants to
spin a PyCmdMessenger instance off onto a non-blocking thread, they should
be able to do so with minimal effort.  I suspect bad things would happen if
they tried to have multiple instances on multiple threads, or if they tried
to access the serial connection on independent threads.)   I also
reconfigured the implementation since I first contacted you so it passes
types as binary strings.  I found it was much more reliable that way.

Thanks again!

Mike

closed 2016-09-14 by thijse