thijse/Arduino-CmdMessenger #11: Sending command via Arduino USB port causes reset

bluespider42, 2016-09-16:
I am using an Arduino Mega 2560 and PyCmdMessenger.
When I send a simple acknowledge type message to the Arduino through the built-in USB/Serial interface it causes a reset of the Arduino.
However if I connect a external USB/Serial convertor directly to the Tx/Rx pins of the Arduino then it works , no problem.
The Arduino USB/Serial port works normally when not using CmdMessenger (ie for looking at debugging messages in tera term).
I can't work out what is causing the reset - or is this possibly a problem with PyCmdMessenger and the way it handles connections?

valkuc, 2016-09-16:
Probably this is you case https://github.com/thijse/Arduino-CmdMessenger/issues/10
However this repo is not related to PyCmdMessenger, if what is written in issue above will not help, try go to https://github.com/harmsm/PyCmdMessenger

harmsm, 2016-09-16:
I maintain PyCmdMessenger and I'm happy to help with the PyCmdMessenger
side if this isn't #10. Just create an issue over there. (I've never seen
this failure mode, but then, I've never plugged in a Mega 2560...)

Mike

> Probably this is you case #10
<https://github.com/thijse/Arduino-CmdMessenger/issues/10>
> However this repo is not related to PyCmdMessenger, if what is written in
issue above will not help, try go to
https://github.com/harmsm/PyCmdMessenger
<https://github.com/thijse/Arduino-CmdMessenger/issues/11#issuecomment-247576154>,
or mute the thread
<https://github.com/

bluespider42, 2016-09-20:
Thanks, I think this probably is my issue - or at least related.
When in #10 you said try setting"DTREnable to true and false" did you mean on the Arduino or the PC side?
I couldn't find a way of doing it in the Arduino software but found a lot of references to setting the reset line manually with a resistor. But I'd rather not have to do that every time I wanted to re-flash it!
If it needs doing on the PC side then I'll move this conversation over to the PyCmdMessenger page as the PyCmdMessenger handles the actual serial connection. I did try adding in a `self.comm.setDTR(value)` into the `arduino.py ` of PyCmdMessenger but I think that was only having an affect after the port was opened (and the reset had occurred).

valkuc, 2016-09-20:
This is related to PC side.

closed 2016-09-20 by bluespider42