thijse/Arduino-CmdMessenger #13: The available()  should return Current!=null;

MikeTumapon, 2016-10-31:
For variable number of arguments, this code will not work:

    while (cmdMessenger.available())
    {
          int curNo=cmdMessenger.readInt16Arg();
          
          // Do something with this number.
    }

It will skip 1 parameter at a time.  Was this function intended for this?

thijse, 2017-01-12:
Honestly, if forgett why I available goes to the next parameter. I do not see samples using it. I will think on your suggestion.

Palatis, 2017-02-05:
I find this "feature" annoying, too...

sigmaeo, 2018-03-13:
Me too.
Instead of:
`if( cmdMessenger.available() )
{
	value1 = cm.readInt32Arg();
	if( cmdMessenger.available() )
		value2 = cm.readInt32Arg();
}`

One have to do it this way:
`ui32 = cm.readInt32Arg();
if( cm.isArgOk() )
{
	value1 = ui32;
	ui32 = cm.readInt32Arg();
	if( cm.isArgOk() )
		ui32 = cm.readInt32Arg();
}`

If I send the command string (command-id=0): 
**0,1,2,3,4,5;**

The first example catches:
value1=2 and 
value2=4

The second example catches:
value1=1 and 
value2=2
as expected.

Palatis, 2026-10-01:
the intended behavior does not comply with other stream library.

usually `available()` only check if any data is available or not, but doesn't consume it (advance read pointer).
thus i (and other people might) assumed a similar behavior.

see https://docs.arduino.cc/language-reference/en/functions/communication/stream/streamAvailable/