thijse/Arduino-CmdMessenger #20: Checksum for wireless connection

MCPC10, 2017-02-02:
I am controlling an rc vehicle with your lib, but I have the problem that I need a crc to check the data (connection over Xbee). But I have no idea how I can make that.

Sorry for my english :)

thijse, 2017-02-05:
Hi, it is very high on the todo list, but I will take it up in the refactoring of the binary protocol. So it may take a while. Alternatively, you can add an additional variable yourself, using e.g. https://github.com/FrankBoesing/FastCRC for Arduino and https://www.codeproject.com/Articles/35134/How-to-calculate-a-CRC-in-Csharp for C#

closed 2017-02-05 by thijse

ElectroLund, 2023-06-05:
the above recommended C# article from Codeproject was deleted.  There are many more though:
https://www.codeproject.com/search.aspx?q=checksum&x=0&y=0&sbo=kw