thijse/Arduino-CmdMessenger #32: CmdMessengerTest failed with Arduino Uno

dontsovcmc, 2017-11-21:
environment:
Arduino UNO
Arduino IDE 1.6.11
Visual Studio 2015.3
Windows 10x64

prepare CommandMessengerTests:
var device = arduinoNano;
PortName  = "COM7"

run CommandMessengerTests (x86, Release)

```
*** Test Summary ****

Some test sets failed!! 
Test sets passed: 2
Test sets failed: 4
```
TestLogFile.txt: https://pastebin.com/wZCkLBT4

dontsovcmc, 2017-11-21:
Usually OK, but sometimes... 

```
 *************************************
 *** Start test-set Waiting for acknowledgments ****
 *************************************

   *** Start test Test sending command and receiving acknowledgment ****

      Sent > 3;
      Not OK: No acknowledgment for command AreYouReady
   *** End test Test sending command and receiving acknowledgment ****
      Warn > Command without attached callback received
```

```
*************************************
 *** Start test-set Clear text data ****
 *************************************

   *** Start test Ping-pong of random plain-text bool values ****

      Received > -1;
      Warn > Command without attached callback received
      Received > 2,Arduino ready;
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      Sent > 7,0,0;
      Warn > Command without attached callback received
      Not OK: Incorrect response
      Warn > Command without attached callback received
      Warn > Command without attached callback received
      OK: Value as expected
      Sent > 7,0,0;
      OK: Value as expected
      Sent > 7,0,0;
....................................
   *** End test Ping-pong of random plain-text bool values ****

   Test failed
   Test parts passed: 100
   Test parts failed: 1
```

```

 *************************************
 *** Start test-set Clear binary data ****
 *************************************

   *** Start test Ping-pong of a command with handpicked binary int16, int32, and double parameters ****
      Warn > Command without attached callback received
      Not OK: Incorrect response
      Sent > 9,/ Ò,}¶³,/,{Qÿ;
      Not OK: No response on ValuePing command
   *** End test Ping-pong of a command with handpicked binary int16, int32, and double parameters ****

   Test failed
   Test parts passed: 0
   Test parts failed: 2
```

adamj537, 2021-09-08:
I am also getting poor test results with similar environment (Windows 10, Arduino UNO, Visual Studio 2019, baud rate 115200).  Adding to confirm the issue still exists.