thijse/Arduino-CmdMessenger #38: readBin() is not "type-length-safe"

sigmaeo, 2018-03-20:
If you use:
uint32_t ui32 = cm.readBinArg();
Then readBinArg() or readBin() will read 4 Bytes from the receiving buffer, ignoring the real length of what is sent and any seperator that may come earlier.
If you send (with ID=1): 
```
1,ABCD;
```
(with A=0x41, B=0x42, C=0x43, D=0x44) you will get as expected: 
0x44434241 = 1145258561.
But if you send:
```
1,A;
```
you don't get 0x41 = 65, instead you will get a random value with 0x41 as LSB, 0x00 after it and 2 more "random" bytes from memory.

This should be enhanced with a test if the length (that slpit_r() sees) is big enough, but for this some more changes are needed.

sigmaeo, 2018-03-20:
I corrected this issue by myself in my fork and so it is included in pull #37.

pauleffect90, 2019-08-07:
If I try to send bin string to the pc app, it works perfectly. the other way around though...
I setup a test command that receives a string and then sends a command with binary args back to the pc.
using numbers, chars etc. works. I input 4, it prints 4. If I input "hello world" as binArg, it prints nothing.
however, if I send a binArg from the arduino to the pc, it works just well.
So Im guessing it had to do with what you kindly explained.