thijse/Arduino-CmdMessenger #49: compile error

sslupsky, 2019-09-14:
Encountered the following error when attempting to compile a sketch using CmdMessenger:

```
Arduino-CmdMessenger/CmdMessenger.cpp:492:9: error: invalid conversion from 'char' to 'char*' [-fpermissive]
  return '\0';

```

The function is defined to return pointer as follows;
https://github.com/thijse/Arduino-CmdMessenger/blob/8c3c4446fa593b8ea96a350c99aa93e7b471fc82/CmdMessenger.cpp#L484

Since the function is returning a char instead of a pointer, this is causing a compiler error.

Should the correct return value be NULL?  ie:

`return NULL;
`

sslupsky, 2019-09-14:
I am using the arduino CLI to compile.

arduino-cli Version: 0.3.7-alpha.preview

sslupsky, 2019-09-14:
The target is a MKRWAN1300 that uses the SAMD21 mcu.

ganjoe, 2019-10-03:
same with stm32duino

adamj537, 2021-09-08:
This library appears to have been written for the AVR compiler, which does not return any errors on this line.  Atmel SAM compilers return an error.

I was able to compile for an Adafruit Feather M0 by changing the offending line to include a cast, as follows:
`return (char)'\0';`

CombiesGit, 2022-03-05:
original:
> return '\0';

hmmm...:
> return NULL;

better:
> return nullptr;

Erhannis, 2024-04-27:
> original:
> 
> > return '\0';
> 
> hmmm...:
> 
> > return NULL;
> 
> better:
> 
> > return nullptr;

Actually...that's a good point.  Looking at the code, I think the author _maybe_ intended to return an empty string - but instead effectively returned NULL.  I'm not sure; I feel like it could go either way.

ccrawford, 2025-01-10:
Using this code on an ESP32, I'm consistently getting crashes when readStringArg encounters the null value condition. 

Actually, the crash happens on a call to CmdMessenger::unescape() that uses the (NULL) result of readStringArg(). Certainly we can write defensive code to check the value of readStringArg and not call unescape if the value is NULL, but I think either readStringArg should return a pointer to an empty string or unescape needs some input validation before trying to dereference an null pointer.

coldfusioncoder, 2025-03-19:
Hitting this issue on MKR Zero, Arduino IDE 2.3.4.