thijse/Arduino-CmdMessenger #52: Problem with SerialTransport

Piotr66, 2020-09-13:
Hi!
My name is Piotr and I want to ask You for a little help with cmdMessenger and CSharp. I’m not sure is this place appropriate, but I decided to try. 
I don’t know what I’m doing wrong, by I can’t help myself with “private SerialTransport _serialTransport;”
My English is not good so maybe I show You on picture:

[image]

[image]

[image]

What I’m doing wrong?

valkuc, 2020-09-13:
You need to add another library CommandMessenger.Transport.Serial

Piotr66, 2020-09-13:
Thank You very much for help. It’s working now.
Earlier I was trying to do this, but I didn’t notice that now is another project and I must create reference for it.

The last problem (for now) left:

[image]

Maybe it’s old example (found in Internet)… 

Btw. I’m looking for simple example to build by myself and analyze how it works.  If You know where I can found it, please let me know.